hpwit/artnetesp32v2 #3: Example code of this library over Ethernet is needed

jagspaul, 2023-07-21:
Please provide example code of this library using dual core, one core to handle Ethernet and another core for rest work, so that that we can get large pixels capacity.

WT32-ETH01 board may be used for this purpose.

I have this board I can go through all the test if you need.

Thanks & Regards
Jags

hpwit, 2023-07-21:
hello
The library is already using two cores in the back end ?
why do you have issues ?
I've never tried it with the WT32-ETH01 . I know of people who used it with ws5500 and it worked flawlessly up to 136 universes
Can you tell me more about you configuration (nb of leds per pins, how many universes, fps ?....)
and maybe the code
Regards
Yves

jagspaul, 2023-07-21:
Tanks for your reply.
I want to drive 16 port each port 512 pixels.

I do not mind to use  ws5500 + esp32 as ethernet board instead WT32-ETH01.

In this regards you are requested to share a sample code of ws5500 + esp32 for ethernet artnet.

Tanks
Jags

hpwit, 2023-07-22:
hello
here it is
https://gist.github.com/hpwit/4effc96f24b5ec5678bcf57879945c1e
 if you can I would increase the number of outputs using the virtual driver in order to increase the native fps of the panel
let me know
which artnet tool are you using?
yves

jagspaul, 2023-07-22:
Hello Sir,

Thanks for your reply.
I have some query as follows.

1. where can I get WebServer_ESP32_W5500 library used in your code.

2. In your code it is found total pixel capacity 24300 (NUM_LEDS_PER_STRIP 2700 , NUMSTRIPS 9). It is just a huge. Now my questions a) what is the frame rate you get with this large pixels panel? b) what SPI speed set to get this large frame from Ethernet? c) Is it really possible to get 24K pixels capacity with ESP32+W5500 over artnet?
3. What artnet sender tools you used for this?

Regarding your questions.....
Right now I don't have hardware for virtual driver but in future i will shift on virtual driver.
I am using Jinx software for artnet sender.

Jags

hpwit, 2023-07-22:
Hello 
This adaptation is from someone who is using my librairies with ws5500. He says that he can manage this at 25fps. The artnet tool is using is a specific Brazilian software. I will look for the name

jagspaul, 2023-07-24:
Hello,
Very nice to you.
I have one through the example code and after the test I could manage 4500
leds at 25fps. Which is just half of my target capacity. I am really
surprised how the other person can manage 136 universes (more than 20k
leds) @25fps.

jags

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

voitnib, 2023-12-03:
> if you can I would increase the number of outputs using the virtual driver in order to increase the native fps of the panel
> let me know

Hi! Can you please share example of code with clockless virtual driver? I'm new to ESP32 and code, I've tried to modify your example, but didn't get any output. Examples from I2SClocklessLedDriver works ok, so I assume I have a mistake in code.

Here is the code: 

`#include <WebServer_ESP32_W5500.h>
#include "Arduino.h"
#include "artnetESP32V2.h"
#include "I2SClocklessVirtualLedDriver.h"
#define DEBUG_ETHERNET_WEBSERVER_PORT Serial
#define _ETHERNET_WEBSERVER_LOGLEVEL_ 3
#define NBIS2SERIALPINS 4
#define STATIC_COLOR_GRB 1
#define NUM_LEDS_PER_STRIP 256 
#define NUM_LEDS (NUM_LEDS_PER_STRIP*NBIS2SERIALPINS*8)
#define CLOCK_PIN 16
#define LATCH_PIN 26
#define NUM_STRIPS 32
#define NB_CHANNEL_PER_LED 3
#define UNIVERSE_SIZE_IN_CHANNEL 512
#define START_UNIVERSE 0 
int pins[NBIS2SERIALPINS] ={0,2,15,12};

uint8_t leds[NUM_STRIPS*NUM_LEDS_PER_STRIP*3];

#define NUMBER_OF_MAC 20
byte mac[][NUMBER_OF_MAC] = {
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x01 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x02 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x03 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x04 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x05 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x06 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x07 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x08 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x09 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x0A },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x0B },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x0C },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x0D },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x0E },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x0F },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x10 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x11 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x12 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0x13 },
  { 0xDE, 0xAD, 0xBE, 0xEF, 0xBE, 0x14 },
};

artnetESP32V2 artnet = artnetESP32V2();
I2SClocklessVirtualLedDriver driver;
void displayfunction(subArtnet *subartnet) {

  driver.showPixels(NO_WAIT,subartnet->data);
    
}

IPAddress myIP(2, 0, 0, 51);
IPAddress myGW(2, 0, 0, 1);
IPAddress mySN(255, 0, 0, 0);

void setup() {
  Serial.begin(115200);

  while (!Serial && (millis() < 5000))
    ;
  driver.initled((uint8_t*)leds,pins,CLOCK_PIN,LATCH_PIN);
  driver.setBrightness(20);
  ESP32_W5500_onEvent();

  ETH.begin(MISO_GPIO, MOSI_GPIO, SCK_GPIO, CS_GPIO, INT_GPIO, SPI_CLOCK_MHZ, ETH_SPI_HOST);
  ETH.config(myIP, myGW, mySN);
  artnet.addSubArtnet(START_UNIVERSE, NUM_LEDS_PER_STRIP * NUM_STRIPS * NB_CHANNEL_PER_LED, UNIVERSE_SIZE_IN_CHANNEL, &displayfunction);

  if (artnet.listen(6454)) {
    Serial.print("artnet Listening on IP: ");
    Serial.println(myIP);
  }

  ESP32_W5500_waitForConnect();
}

void loop() {
  vTaskDelete(NULL);
}`

I'm getting messages that W5500 connetcted to ethernet and I see a device on my router, but nothing happens when I try to send artnet data to 2.0.0.51