hpwit/artnetesp32v2 #4: ArtPoll node discovery

jagspaul, 2023-08-25:
ESP32 boards running the artnetesp32v2 connected with artnet network. however Madrix software does not find any "artnet nodes" on my network. I think in artnetesp32v2 ArtPoll node discovery is not supported.

Can you help me how to add support of ArtPoll node discovery?

Thanks 
Jags

hpwit, 2023-08-26:
Hello
I will have a look at this. 
can I ask how many universes you are driving with the driver ?

hpwit, 2023-09-03:
Hello
At which framerate are you using it ? is it smooth ?
which ethernet library are you using ? ws5500 or other .
Sorry for the questions but  I am interested when other people use my library and the use of art netesp32v2 and ethernet bring a lot of questions.
I am just back from holidays I will look at that asap

jagspaul, 2023-09-03:
My requirment is 25 FSP and i found very satisfactory results with Ethernet
as well as wifi.

louisnorwood, 2023-10-25:
+1 for this feature. I'm building a bunch of controllers using esp32 and wiz5500. Each one is only doing a few universes(4-8), After much testing this library is so awesome and fast, I still want to use it so I never miss a frame. 
Would be so helpful to have poll feature, so the devices are discoverable. This would make setup and commissioning so much easier.

hpwit, 2023-10-26:
By popular demand ;) I will work on this feature then. 
I need to finalize something on the leds driver. 
And I will implement it. I will try to work on it next week.

louisnorwood, 2023-11-20:
Hello, any chance you could start thinking about this, I had a bit of a read of the specto see if I could assist, but confused the hell out of me. 
Seems the discovery packet contains loads of information about the controller, such as universe IDs, protocol versions and other stuff. 
Really all that isn't necessary for my purpose, I just wanted it to be found in Resolume so I didn't have to keep a massive list of fixed IPs, and when a node went down I could see it wasn't present in the device list making it easier to configure and debug the installation.
Thanks, Louie

hpwit, 2023-11-20:
Hello
Unfortunately I had a quick look at it and did not really tackle the point.
Indeed it seems a lot of information needs to be present. I did not know Resolume had discovery functionalities
I will try to get to it asap

louisnorwood, 2023-11-20:
Hello,
Really I just imagined that it would show up as a device, it's just so the IP is visible and you can see the device in the list of targets.
I know it's a bit fo a hacky implementation not to have it as a fully populated device, but what's important is that the software knows there is an enpoint to connect to, that would make life much easier.

Thank you

hpwit, 2023-11-20:
I have found a way I think I need to implement it

hpwit, 2023-11-21:
Hello it's done on the dev branch
it's nor on the main because it's really dirty for now
you have a new function
setNodeName(String s)

artnetESP32V2 artnet=artnetESP32V2();

artnet.setNodeName("yves");

let me know

louisnorwood, 2023-11-22:
You seemed to have changed the callback function in the Dev branch, so it now just returns data rather than subartnet.

I was using the 'subartnet->subArtnetNum' to extract the Universe, so I could use the same callback function for all my universes and map them to the correct pixels. Is this now noit possible, or is there some way I can still get the Universe (SubartnetNum) from within the callback?

jagspaul, 2023-11-23:
Does it mean that now  artnetESP32V2 ( dev branch  )  supports  ArtPoll
node discovery??

hpwit, 2023-11-23:
Oh indeed I had change this !!! @louisnorwood why do u need the universe number ?  I will add the capability to have both. 
Before using Ethernet I need to had something I will try to do it today clean the code and merge into main. Happy that at least with wifi it's working also at yours. @jagspaul yes the dev branch has artpool discovery

louisnorwood, 2023-11-23:
@hpwit I use the Universe number to simplify my callback routine. 
I output 2 Universes on 8 different pins from a single callback routine. This reads the subartnetNum and uses this to point to the correct Array for that Pin, without that number, I would need 8 callback routines in my code. 

I would really like to try and get that functionality working on your I2S clockless driver, but I still want to be able to manipulate the data, as for someapplications I want to change the data, so having a for loop where I pass the recevied data into the display array is very useful for me, but with your clockless driver it just hands it over as a pointer, which meant I couldn't do any of my manipulation between RX and output.

hpwit, 2023-11-23:
You're doing a mapping ?
I will give you back that functionality but I think the driver as the capabilities to simplify what you do greatly 
Would you mind sharing the code of your 'transformation' to the 8 strips ?
Yves

hpwit, 2023-11-23:
hello
new dev version
to get back the subartnet  add
#define ARTNET_CALLBACK_SUBARTNET before the include

to use ethernet for the artpoll
do this before the include
#define ARTNET_CONNECT_MODE ARTNET_ETHERNET
if your ethernet library is Ethernet 
or
#define ARTNET_CONNECT_MODE ARTNET_ETH
if your library is ETH
@louisnorwood  let me know

I think I will also have a revamp of the library I think some improvement can de done

louisnorwood, 2023-11-23:
@hpwit 

This is the code I am using in my callback to map Universes to different FastLED Arrays
`void callbackfunction(subArtnet  * subartnet){

  int Arrayindex = subartnet->subArtnetNum % 8; 
    for (size_t pixel = 0; pixel < LEDS_PER_UNIVERSE; pixel++) {
            size_t idx = pixel * 3;
            leds[Arrayindex][pixel].r = subartnet->data[idx + 0];
            leds[Arrayindex][pixel].g = subartnet->data[idx + 1];
            leds[Arrayindex][pixel].b = subartnet->data[idx + 2];
    }
  }`

Yes I'm helping a friend who is making triangular panels, but the software only supports squares, so I'm throwing away artnet pixels to get the map working correctly on the Arduino, pretty proud of that idea, seems to work well so far.

I'll give your updates a test now

louisnorwood, 2023-11-23:
@hpwit I can't get this SUBARTNET callback to work.

I get this error:
invalid conversion from 'void (*)(subArtnet*)' to 'void (*)(uint8_t*)' {aka 'void (*)(unsigned char*)'} [-fpermissive]

```
#define ARTNET_CALLBACK_SUBARTNET
#include "artnetESP32V2.h"
//---------------------------------------------------------------------------

void callbackfunction(subArtnet  * subartnet){
  int Arrayindex = subartnet->subArtnetNum % 8; 
    for (size_t pixel = 0; pixel < LEDS_PER_UNIVERSE; pixel++) {
            size_t idx = pixel * 3;
            leds[Arrayindex][pixel].r = subartnet->data[idx + 0];
            leds[Arrayindex][pixel].g = subartnet->data[idx + 1];
            leds[Arrayindex][pixel].b = subartnet->data[idx + 2];
    }
  }
```

hpwit, 2023-11-23:
Sorry reload the new dec branch please 
Let me know

louisnorwood, 2023-11-23:
@hpwit Ah damn, not sure wha't gone wrong, but I jsut have boot looping with the latest dev branch now whenever a packet is received, only happenening when I try to use #define ARTNET_CALLBACK_SUBARTNET
```

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d2c51  PS      : 0x00060030  A0      : 0x800d61e9  A1      : 0x3ffd94a0  
A2      : 0x3ffceb04  A3      : 0x00000000  A4      : 0x3ffd72b4  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x80000001  A8      : 0x3ffc39c8  A9      : 0x00000000  
A10     : 0x00000154  A11     : 0x00000000  A12     : 0x73f56948  A13     : 0x3ffd943c  
A14     : 0x3ffc4710  A15     : 0x3ffd943c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d2c4c  LEND    : 0x400d2c71  LCOUNT  : 0x00000153  

Backtrace: 0x400d2c4e:0x3ffd94a0 0x400d61e6:0x3ffd94c0

ELF file SHA256: 9488ad72990887f7

```

hpwit, 2023-11-23:
humm strange I will have a deep look at it
something stupid from my side I guess
regarding your
void callbackfunction(subArtnet  * subartnet){
  int Arrayindex = subartnet->subArtnetNum % 8; 
    for (size_t pixel = 0; pixel < LEDS_PER_UNIVERSE; pixel++) {
            size_t idx = pixel * 3;
            leds[Arrayindex][pixel].r = subartnet->data[idx + 0];
            leds[Arrayindex][pixel].g = subartnet->data[idx + 1];
            leds[Arrayindex][pixel].b = subartnet->data[idx + 2];
    }
  }
it could be simplified by
void callbackfunction(subArtnet  * subartnet){
  int Arrayindex = subartnet->subArtnetNum % 8; 
memcpy(leds[Arrayindex],subartnet->data, LEDS_PER_UNIVERSE*3);
}

I don't understand why you need to copy all the universes into a led array. because you're copying the entire universe.
SO I guess you have 8 subartnet definition ?

hpwit, 2023-11-24:
reload the dev branch
 and do tthis
void callbackfunction(void  * param){
  subArtnet * subartnet = (subArtnet *)param;
  int Arrayindex = subartnet->subArtnetNum % 8; 
    for (size_t pixel = 0; pixel < LEDS_PER_UNIVERSE; pixel++) {
            size_t idx = pixel * 3;
            leds[Arrayindex][pixel].r = subartnet->data[idx + 0];
            leds[Arrayindex][pixel].g = subartnet->data[idx + 1];
            leds[Arrayindex][pixel].b = subartnet->data[idx + 2];
    }
  }

crazy stuff about the compiler made me do this but anyway
let me know

louisnorwood, 2023-12-09:
Still no love sadly, getting this error when using the latest

invalid conversion from 'void (*)(void*)' to 'void (*)(subArtnet*)' [-fpermissive]

Happy to just work with the data structure and manage my multiple callbacks, so not really an issue if it's a pain to fix, easy enough to work around

hpwit, 2023-12-11:
Hello 
That is strange has I have tried it. 
I will look at it deeper to try to understand. 
How many subartnet are you using 
Yves

louisnorwood, 2023-12-24:
Hey Yves,
Your suggestion I do have working SubArtnet format now receiving.

Still having some issues with the latest branch, details all below. 

//------ SUBARTNET CALLBACK - WORKING --------------------------------------------------
This Now Works, when receiveing 2 or more unverses

#define ARTNET_CALLBACK_SUBARTNET

void callbackfunction0(void * param){
  subArtnet * subartnet = (subArtnet *)param;
  Serial.printf("--call back Universe:%d size:%d\n",subartnet->subArtnetNum,subartnet->len);
  memcpy(leds[0], subartnet->data, CHANNELS_PER_PIN);
}

//---- DATA CALLBACK - NOT WORKING ----------------------------------------------------
Can get compilation, but not getting the data correctly with this (when using data array), presuming it's the format of this:

void callbackfunction0(void * param){
unsigned char * data = (unsigned char *)param;
  memcpy(leds[0], data, CHANNELS_PER_PIN);
}

// ------SINGLE UNIVERSE - NOT WORKING -----------------------------------------------------
When setting the Callback Length to 170 LEDs (510 channels)

The last coupe of commits in the Dev Branch have broken the single Universe operation, the main branch still supports using 510 Channels, but the Dev Branch just doesn't trigger the callback, I presume it's lenght related, have tried using 512 and other things, but never get a callback when the incoming data is under 2 Universes. 

Not sure how to debug this, as I just don't get anything received, would be useful, as I have several devices that I am using single universe callbacks, but am currently padding the data to 2 universes which makes it work correctly.

Was gonna raise a new ticket, but I thought worth updating you on all my finds in this thread.

Happy Christmas to you.
Louis

hpwit, 2023-12-25:
hello
that line
memcpy(leds[0], subartnet->data, CHANNELS_PER_PIN); is not gonna work you are writing somewhere in the memory and I am not sure it will compile at all. leds[0] is a CRGB type 
it should be memcpy(&leds[0], subartnet->data, CHANNELS_PER_PIN) or memcpy(leds, subartnet->data, CHANNELS_PER_PIN);
Can I see your code because I don't think you do't need 8 subartnet

louisnorwood, 2023-12-25:
Happy Christmas to you and your family too.
Sorry for the confusion, this code does work.
I make an array of arrays the first is to support each different pin, LEDs[0][0] is the crgb value for the first pixel on the first pin.

The two issues I have is not knowing how to define the data type array in the latest, as I get the callback type void (as i did with the subartnet type (before your compiler trick above).
And the second is not being able to receive a single universe (170 LEDs/510 channels). This no longer triggers the callback is it did in the main branch.
I am successfully using this dev library, and it's performing well, I currently just send over one  universe of data to a small node to trigger the callback

hpwit, 2023-12-25:
Hello Louis 
I will have a look  at it. 
Rgds
Yves

louisnorwood, 2023-12-28:
> Hello Louis I will have a look at it. Rgds Yves

Hello Yves, I had a look into it for you, had to learn how to use ESP_LOGV, and glad I did.

I believe this is to do with the `frame_disp` variable. 

Whenever there is only one Universe, the handleUniverse() function never goes into the `if (current_uni == previousUniverse + 1)` section, and therefore never resets the frame_disp variable to true, which means it does not ever add it to the queue.

I don't fully understand the magic in this library, from chasing through debug messages, I am confident that this is the issue. I hope this helps.

Cheers,
Louie

louisnorwood, 2023-12-28:
I went a bit deeper, and got something working, although it's not elegent so you would probably want to rewrite.

Basically, with 1 universe it never gets to `if (current_uni == previousUniverse + 1)`, so it never sets the memset or `frame_disp=true;`

So to handle this, I duplicated that check function just before the return in `if(current_uni == startUniverse)`. Essentially chcecking if this is the end_universe as well as the start universe. I tried to take that end check out of the if to avoid duplication, but just got bootlooping, so this is the best hack I can muster

So from line 338 my handleUniverse function looks like this:
```
        previousUniverse = current_uni;
        memcpy(offset, payload + ART_DMX_START, nbDataPerUniverse);
        tmp_len = (length - ART_DMX_START < nbDataPerUniverse) ? (length - ART_DMX_START) : nbDataPerUniverse;
//--- NEW BIT --
        if (current_uni == endUniverse - 1)
            {
                // nb_frames++;
                data = offset;
                currentframenumber = (currentframenumber + 1) % nbOfBuffers;
                offset = buffers[currentframenumber];
                memset(data+_nb_data+3,10,3);
                frame_disp=true;

            }
//--- END OF NEW BIT --
        return;
```

This Works with one Universe, also works with 2 Universes. I guess you will rewrite this cleaner for this edge case, and I thought this time I spent would assist you in this fix.

Have a great day.
Louis

hpwit, 2023-12-29:
hello Louis
back home finally, indeed  I had overseen this case. thank you for noticing
this. I will look at your fix and see how to integrate it the best way
possible.
this function is a bit of a mess I need a way to simplify it anyway
question why are you using 8 subartnet instead of one big subartnet

Yves

hpwit, 2024-01-06:
Hello
I have published a new version and also merged it into  the main
it should deal with the 1 universe issue you had
rgds
Yves

jagspaul, 2024-06-13:
I tested new version artnetesp32v2 with artnet network. however Madrix software does not find any "artnet nodes" on my network. I think  artnetesp32v2 ArtPoll node discovery still not working.

However if I set manual IP address to Madrix software for artnet protocol then also artnetesp32v2 is not working. artnetesp32v2 Artnet protocol is not compatible with  Madrix software.

Pls look in this matter.

Thanks
Jags

hpwit, 2024-06-13:
Hello 
The artnet pool discovery is working with arena. I will try to download Madrid software to see why it's not discoverable. 
Do you know if they have a try version ?

jagspaul, 2024-06-13:
Yes Madirx provides trial version.

https://www.madrix.com/support/download

hpwit, 2024-06-13:
I will need to take out my windows laptop ;)

louisnorwood, 2024-06-14:
I think Madrix is closed source, so I didn't think it's possible to use 3rd party hardware. but I may be wrong

jagspaul, 2024-06-15:
Madrix software is close source but the protocol used by mardix is artnet which is open. So madrix software is compatible with any 3rd part control support artnet.