hpwit/artnetesp32v2 #5: Library breaks using 1020 leds * 8 strips 48 universes

Xdevel0per, 2024-01-19:
When I try the library using 8 strips (8 pins) each one using 1020 leds combining i2s_clockless_led_driver

In verbose log it says impossible to create buffer 7
And never calls the display function

I want to know what's the maximum limits of this library
Thx 🙏❤️

Xdevel0per, 2024-01-19:
I2s library manages well this amount of leds on it's own
But the artnet_v2 library doesn't

And also I want to know more about when to use the full dma buffer

Sorry bud for interrupting
I appreciate it and I appreciate your work 🙏🤍

hpwit, 2024-01-20:
Hello
That should not be the problem. At which fps are you sending the data ?
Because 1020 leds will be running at 30fps max so it could be an issue. 
Can you share the code please ?
Let me add couple of more verbose for you to find the why. I will push a new version soon.

Xdevel0per, 2024-01-20:
I tried sending the data at 10 , 20 , 24 , 30 , 40 , 60 fps
None of them worked

When I start the esp it logs in debug that it's creating different buffers then it says
Impossible to create buffer 7

I'll comment the code and the logs in just one hour because I'm not home ❤️🙏

hpwit, 2024-01-20:
Hello
The fact that it can't create all the buffers is not an issue it will not block the program 
What is that line ?

ESP32_ENC_waitForConnect();

If it's for the connection it should be before the if (artnet.listen(6454)) ...
 Can you post the full serial output of the verbose please.

Xdevel0per, 2024-01-20:
Hello
Thank you for ur effort to help and I'm sorry for interrupting you
The code I have is the same code you posted about working with the library via Ethernet
The only thing I changed was the leds count and the library
Webserver_esp32_w5500
To
Webserver_esp32_enc

I'll post you the serial log in a minute
Thanks again ❤️🙏

Xdevel0per, 2024-01-20:
heres the output

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, [    11][V][I2SClocklessLedDriver.h:1067] __initled(): [I2SClocklessLedDriver] xdelay:31
[    30][V][WiFiGeneric.cpp:438] _arduino_event_cb(): Ethernet Started
[    30][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 18 - ETH_START

ETH Started
[    83][I][artnetESP32V2.cpp:233] _initialize(): [ARTNETESP32] Initialize subArtnet Start Universe: 0  end Universe: 47, universes 48
[    84][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 0
[    91][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 1
[    99][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 2
[   107][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 3
[   114][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 4
[   122][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 5
[   130][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 6
[   138][I][artnetESP32V2.cpp:185] createBuffers(): [ARTNETESP32] impossible to create the buffer 7
[   146][I][artnetESP32V2.cpp:197] createBuffers(): [ARTNETESP32] nb total of buffers:7
[   154][I][artnetESP32V2.h:488] _udp_task_start(): [ARTNETESP32] QUEUES CREATED
[   161][I][artnetESP32V2.h:376] _udp_task_subrarnet(): [ARTNETESP32] Start listening task
[   161][V][artnetESP32V2.h:424] _udp_task_subrarnet_handle(): [ARTNETESP32] _udp_task_subrarnet_handle set on core 1
[   169][I][artnetESP32V2.h:381] _udp_task_subrarnet(): [ARTNETESP32] subArtnet nr:0 start universe:0 Nb Universes: 48
artnet Listening on IP: 192.168.1.10
[ 20033][V][WiFiGeneric.cpp:431] _arduino_event_cb(): Ethernet Link Up
[ 20033][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[ 20034][V][WiFiGeneric.cpp:446] _arduino_event_cb(): Ethernet got newip:192.168.1.10
ETH Connected
[ 20054][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[ 20054][D][WiFiGeneric.cpp:1138] _eventCallback(): ETH IP: 192.168.1.10, MASK: 255.255.255.0, GW: 192.168.1.1
ETH MAC: EC:94:CB:6C:F5:73, IPv4: 192.168.1.10
FULL_DUPLEX, 10Mbps
E (20081) enc28j60: no mem for receive buffer

hpwit, 2024-01-20:
I see that the eth driver do not have enough memory. 
Got to the file artnetesp32v2.cpp 

Line 23 
#define NB_MAX_BUFFER 10

Change 10 to 3

And retry 

Let me know

Xdevel0per, 2024-01-21:
I'll try as soon as possible and I'll let u know

Xdevel0per, 2024-01-22:
Sorry I was so ill
I'll try today and I'll give you my output

Also another important question

In industrial artnet controllers we have the ability to wire up the master controller to slaves using Ethernet
So we have an Ethernet wire from to the master and another from the master to the slave and from slave to slave so each one has an input and an output
Is there's anyway to do that ?

hpwit, 2024-01-22:
Hello,
I hope you're doing better.
Let me know if this solves your issues
answer to your questions;
_>> Q: heres the code and i want to know if its even possible to drive 2040 leds x 8 ports or 1020 leds x 16 ports_
A: the more pins the better because your refresh rate will be higher. if you're not against a bit of soldering I would suggest to migrate to the virtual-in driver with will allow you to have 32 strips of 510 leds using only (6 pins of the esp32 1 clock 1 latch 4 data pins)

_>>Q: So we have an Ethernet wire from to the master and another from the master to the slave and from slave to slave so each one has an input and an output
Is there's anyway to do that_ 
A: I don"t know if it can be done with an esp32 because that would mean your master is also an ethernet router (unless in that case it's not real TCP IP but another protocol just using ethernet as a medium) do you have some product in mind so I can  look it up

Xdevel0per, 2024-01-22:
think with me 
What if we didn't use Ethernet for output
What's the best protocol we can transfer through the data from one esp32 to the other
Maybe serial or i2c
And we can still use Ethernet as a port but not as a protocol

Xdevel0per, 2024-01-22:
I've tried the new change about the nb of buffers
It's working but it's soo laggy
I'm getting about 0.3fps from 48 universes
And I even tried 6 max buffers
Still the same results

Xdevel0per, 2024-01-22:
[image]

[image]

[image]

and also the artnet poll isnt working

Xdevel0per, 2024-01-22:
heres the full log

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, [    11][V][I2SClocklessLedDriver.h:1067] __initled(): [I2SClocklessLedDriver] xdelay:31
[    30][V][WiFiGeneric.cpp:438] _arduino_event_cb(): Ethernet Started
[    30][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 18 - ETH_START

ETH Started
[    83][I][artnetESP32V2.cpp:233] _initialize(): [ARTNETESP32] Initialize subArtnet Start Universe: 0  end Universe: 47, universes 48
[    84][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 0
[    91][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 1
[    99][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 2
[   107][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 3
[   114][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 4
[   122][I][artnetESP32V2.cpp:204] createBuffers(): [ARTNETESP32] Creation of  buffer 5
[   130][I][artnetESP32V2.cpp:210] createBuffers(): [ARTNETESP32] nb total of buffers:6
[ 28033][V][WiFiGeneric.cpp:431] _arduino_event_cb(): Ethernet Link Up
[ 28033][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 20 - ETH_CONNECTED
[ 28034][V][WiFiGeneric.cpp:446] _arduino_event_cb(): Ethernet got newip:192.168.1.10
ETH Connected
[ 28054][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 22 - ETH_GOT_IP
[ 28054][D][WiFiGeneric.cpp:1138] _eventCallback(): ETH IP: 192.168.1.10, MASK: 255.255.255.0, GW: 192.168.1.1
ETH MAC: EC:94:CB:6C:F5:73, IPv4: 192.168.1.10
FULL_DUPLEX, 10Mbps
[ 28137][I][artnetESP32V2.h:488] _udp_task_start(): [ARTNETESP32] QUEUES CREATED
[ 28138][I][artnetESP32V2.h:376] _udp_task_subrarnet(): [ARTNETESP32] Start listening task
[ 28138][V][artnetESP32V2.h:424] _udp_task_subrarnet_handle(): [ARTNETESP32] _udp_task_subrarnet_handle set on core 1
[ 28141][I][artnetESP32V2.h:381] _udp_task_subrarnet(): [ARTNETESP32] subArtnet nr:0 start universe:0 Nb Universes: 48
artnet Listening on IP: 192.168.1.10

hpwit, 2024-01-25:
Hello sorry for the delay
I have been asking around,  which board are you using ? 
Someone I know is using the THE01 and does not have any issues
Reagrds

Xdevel0per, 2024-01-25:
Do you mean the esp01 ?

hpwit, 2024-01-25:
No the esp eth-01 it's an esp32 with Ethernet. 
You're using wich ethernet add on ?

Xdevel0per, 2024-01-25:
I'm using the enc28j60

closed 2024-02-09 by Xdevel0per