hpwit/artnetesp32v2 #7: error: 'powf'

SevenAntares, 2024-04-21:
Hi, I'm trying to use your great library. But I have this error.

- File: artnetdisplay.ino(example)
- Arduino with 2.0.6 core library
- IDE: Arduino IDE
- Board DevModule
- I2SClocklessLedDriver ver 1.3

In file included from C:\Users\Portatil-Trabajo\Documents\Arduino\libraries\Artnetesp32v2\examples\artnetdisplay\artnetdisplay.ino:9:
c:\Users\Portatil-Trabajo\Documents\Arduino\libraries\I2SClocklessLedDriver\src/I2SClocklessLedDriver.h: In member function 'void I2SClocklessLedDriver::setBrightness(int)':
c:\Users\Portatil-Trabajo\Documents\Arduino\libraries\I2SClocklessLedDriver\src/I2SClocklessLedDriver.h:278:19: error: 'powf' was not declared in this scope
             tmp = powf((float)i / 255, 1 / _gammag);
                   ^~~~

exit status 1

Compilation error: exit status 1

hpwit, 2024-04-21:
Hello
Can you add 
#include <math.h>

At the beginning of your arduino script ?
Yves

SevenAntares, 2024-04-21:
Thank you!! Now it works!

hpwit, 2024-04-21:
Great  !!!
Does it work well I mean is it fluid ?

Thank you can you tell me more about your build, (if you can) I am always eager to know what people do realised baser on my libraries

hpwit, 2024-04-22:
thank you for  sharing 
you can just put the debug level to info you'll win in fluidity
let me know if you need some more help