hpwit/artnetesp32v2 #8: 40 leds not show

SevenAntares, 2024-05-08:
I continue testing with your library.
2 strips of 300 LEDs. connected independently of each other
The first strip connected to pin 23
The second strip connected to pin 12
In Resolume I send 4 universes
 -170
 -130
 -170
 -130
I have also tried other configurations:
 -170
 -170
 -170
 -130

The problem is that the second strip never shows the first 40 LEDs. why?
[image]

hpwit, 2024-06-13:
hello
Can I see the code ?
Regards
Yves

SevenAntares, 2024-06-14:
Sorry, I solved it. Use your example code, but change ;

 #define NUM_LEDS_PER_STRIP 300
  for:
 #define NUM_LEDS_PER_STRIP 340

and it worked