hpwit/artnetesp32v2 #9: Frame Drops

catvvvveasel, 2024-05-19:
Hi, I'm trying out the library on an lolin d32 pro board, with 8 outputs with 1 universe per output but the output is very juddery, its almost acceptable at 120 pixels per output. 
I'm using the exmple in the library but with different pin numbers 
I've got a static IP, but I also had issues when it was dynamic as well. Is it the board do you think?

`

#define NUM_LEDS_PER_STRIP 170
#define NUMSTRIPS 8
#define NB_CHANNEL_PER_LED 3 //Should be 4 if your sending tool is sending RGBW
#define COLOR_GRB
#define UNIVERSE_SIZE_IN_CHANNEL  (170 * 3)  //here we define a universe of 170 pixels each pixel is composed of 3 channels
#define START_UNIVERSE 0
//#define USE_PIXELSLIB
#define UNIQUE_SUBARTNET

#include "I2SClocklessLedDriver.h"
#include "Arduino.h"
#include "WiFi.h"
#include "artnetESP32V2.h"
//here we have 3 colors per pixel

// Set your Static IP address
IPAddress local_IP(10, 0, 1, 30);
// Set your Gateway IP address
IPAddress gateway(10, 0, 1, 1);

IPAddress subnet(255, 255, 0, 0);

const char * ssid = "*****";
const char * password = "*****";

int pins[8]={26,15,2,0,4, 5, 3,18};
//      q     01 02 03 04 05 06 7 8 9 10 11 12 13 14 15 16
// int pins[16]={13,12,14,27,26,15,2,0,4, 5, 3,18,19,21,22,23};

artnetESP32V2 artnet=artnetESP32V2();
I2SClocklessLedDriver driver;
  void displayfunction(void *param){
    subArtnet *subartnet = (subArtnet *)param;
    //  Serial.println("show");
     driver.showPixels(NO_WAIT,subartnet->data);
}

void setup() {
  // put your setup code here, to run once:
Serial.begin(2000000);
 // Configures static IP address
  if (!WiFi.config(local_IP, gateway, subnet)) {
    Serial.println("STA Failed to configure");
  }
  

    /* if you do not use the driver for other thing than artnet you do not
     * need to put a led buffer. If you need one you can use it. In this example It's not needed
     * as the subArtnet contains the buffer to be displayed.
     */
   driver.initled(NULL,pins,NUMSTRIPS,NUM_LEDS_PER_STRIP);  
    driver.setBrightness(20);
       
 WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
Serial.println("OK");
    /* We add the subartnet reciever. Here we only need one
     * addSubartnet (Start universe,size of data,number of channel per universe,callback function)
     * NUM_LEDS_PER_STRIP * NUMSTRIPS is the total number of leds 
     * NUM_LEDS_PER_STRIP * NUMSTRIPS * NB_CHANNEL_PER_LED is the total number of channels needed here 3 channel per led (R,G,B)
     */ 
    artnet.addSubArtnet(START_UNIVERSE ,NUM_LEDS_PER_STRIP * NUMSTRIPS * NB_CHANNEL_PER_LED,UNIVERSE_SIZE_IN_CHANNEL ,&displayfunction);
    artnet.setNodeName("Arnet Node esp32");
     
    if(artnet.listen(6454)) {
        Serial.print("artnet Listening on IP: ");
        Serial.println(WiFi.localIP());
       
    }
}

void loop()
{
  vTaskDelete(NULL);
}`

catvvvveasel, 2024-05-19:
Ah just tried sending from Resolume instead of vvvv which I usually use and its fine! So I'll have to look at that, sorry for the bandwidith wastage!

closed 2024-05-19 by catvvvveasel

hpwit, 2024-05-20:
Hello
Thank you for using my libraries ;)
Can you tell me what was the original program you were using ?
Happy to hear that it works with resolume. 
I have also noticed that if your computer is plug to a router via the Ethernet it works better. 
Do not hesitate if you have other questions. 
Can you tell me more about your build?

catvvvveasel, 2024-05-20:
Thank you for making them!
I was using vvvv gamma vvvv.org
Which I have used to map hundreds of universes, so didn’t even consider that it could be the problem! I can only think that it sends all the universes too quickly maybe, but in resolume I can send less than all of the and the just the sent ones update, I’ve got a busy week but I’ll look into after the weekend.
Current project is for a small festival where I’m using home made led sticks, but ongoing it’s for a project called trees that we toured earlier in the year and needs everything wireless and battery operated!
https://www.thingsthatgoonthings.org/trees
I was using a different library for that (but still had stutters!) so I’m seeking a solution, which might be using resolume now! I had tried yours back then but couldn’t get it to compile, works fine now!

hpwit, 2024-05-20:
Oki nice  maybe I will have a look at the software to see what could be the issue. 
How are the performances with the library and driver ?

hpwit, 2024-05-20:
Nice to hear !! Let me know if you need further help