hpwit/artnetesp32v2 #12: IPv6Address.h not found

bygerardvisuals, 2025-01-20:
Hello,

First of all, thanks a lot for the library and your hard work. I've seen your demo videos and are amazing.

I'm trying to upload the code into an ESP32 but having some issues with multiple directories. (IPv6Address.h, uip-conf.h, uip.h, uipethernet-conf.h, uipopt.h)

I've been reading on the internet that some of them are not anymore included with the ESP32 library and configurations and the kind of error I receive with all of them is more or les this one:

```
In file included from /private/var/folders/tq/rz1mh3f11xn0350s53xzhq1h0000gn/T/.arduinoIDE-unsaved2025020-21948-pyicmu.k9fu9/sketch_jan20a/sketch_jan20a.ino:13:
/Users/user/Documents/Arduino/libraries/WebServer_ESP32_W5500/src/WebServer_ESP32_W5500.h:77:4: error: #error This code is designed to run on ESP32 platform! Please check your Tools->Board setting.
   77 |   #error This code is designed to run on ESP32 platform! Please check your Tools->Board setting.
      |    ^~~~~
In file included from /private/var/folders/tq/rz1mh3f11xn0350s53xzhq1h0000gn/T/.arduinoIDE-unsaved2025020-21948-pyicmu.k9fu9/sketch_jan20a/sketch_jan20a.ino:16:
/Users/gerard/Documents/Arduino/libraries/artnetesp32v2-main/src/artnetESP32V2.h:5:10: fatal error: IPv6Address.h: No such file or directory
    5 | #include "IPv6Address.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

I need to control a 60 pixels width x 20 pixels height panel WS2028B. The board I'm using is an ESP32 C3 SuperMini and would like to control it from MadMapper.

Is there any updated library or anything I'm missing to use now in 2025? I saw that some of the videos are not that old (from 2024) so probebly you've encountered this issue before too.

Thanks a lot in advance for your help!

All the best,
Gerard

hpwit, 2025-01-20:
Hello
Try to comment the line
And see if it compiles
Regards 
Yves

bygerardvisuals, 2025-01-20:
Hello,

Thanks for your really fast reply! Tried and no, it does not compile :( 

`//#include "artnetESP32V2.h"`

This is the one I commented and now have multiple more errors

hpwit, 2025-01-20:
I did not pay attention that you were using a c3. I have never tested that on a c3. 
Sorry

bygerardvisuals, 2025-01-20:
No worries! It does neither compile on a regular one. I've tried to use a ESP32 devkit and same issue. Which one do you use?

bygerardvisuals, 2025-01-20:
Also tried commenting the `//#include "IPv6Address.h"` line on the artnetESP32V2 library and still the same. It shows multiple errors because it cannot find the libraries.

catvvvveasel, 2025-01-20:
https://github.com/hpwit/artnetesp32v2/issues/10

This sounds like this issue, have you tried rolling the board library back to the one mentioned?

bygerardvisuals, 2025-01-20:
Yes is totally that! Thanks a lot!! It seems that will not work with the c3 because it's too new and was not included when they've launched that version but it compiles for the ESP32 DevKit.

Huge thanks again!