hpwit/artnetesp32v2 #13: Open-source ESP32/W5500-based Artnet Controller

mdethmers, 2025-03-12:
Hey there!

Ive started a new project based on your Artnetesp32v2 project. It is a fully open source esp32  based wired artnet node enjoying all the great things this library brings, as well as added support wired connection through a w5500, and added debugging and information display through a status led and oled screen. Also added is a open-source PCB design to mount all the components that make connecting modules and strips a breeze.

However, I have found 1 issue that I am not able to figure out;

The output brightness of this artnetesp32v2 library on the led strips I am using (ws2811) does not match the output brightness of the neopixel library I am using to generate some debugging test patterns.

There is a notable difference  in brightness when using artnet data from resolume (brightness 100 checked) compared to when I use the adafruit neopixel library to control the led strips. Would you have any idea why? 

I initialise the neopixel strips like this; 

`Adafruit_NeoPixel strip1(NUM_LEDS_PER_STRIP, 12, NEO_GRB + NEO_KHZ800);`

Which specifies the use of the 800khz these strips usually run on. I suspect it might have to do something with this, but I am not sure. Any help is appreciated!

catvvvveasel, 2025-03-13:
setBrightness(int brightness):
This function sets the default brightness for 0->255

Have you tried 255? Looks like a useful addition having a web gui and wired connection

hpwit, 2025-03-13:
Hello 
Thank you for your interest in my library and great to build something out of it. 
Normally my library do not touch the brightness of the leds but just pushes the data from artnet to the led buffers. 
Can you tell me more about what your are experiencing?

hpwit, 2025-03-17:
Hello I wrote an email to you 
Let's talk when you have time